Slimbook-Team/slimbookbattery #20: Window too tall, can't save changes

Pizzacutter-at-tty3, 2021-10-08:
Hi, the window where preferences are set is too big to be displayed properly at 1366x768 displays. Buttons on the bottom are clickable, but I can't see what they say until I move the windows with a shortcut key (XFCE 4.14). Although, the most important issue is that the accept button does not do anything and it's impossible to save my changes. The old window worker better, this also seems a bit overstyled for my taste.
Version that I have installed is 4.0.2

madmanRB, 2021-10-09:
Yeah I have the same issue, the UI cant be resized and is unusable...

what a terrible UI choice! They need to go back to the old UI because that could get resized easily.
I am reverting to the old version personally.

Pizzacutter-at-tty3, 2021-10-11:
The button doesn't do anything at all.

PacoMoraMartin, 2021-10-21:
@Freeplayg can you open it in terminal and share the output? Why did you choose this issue, is it related to your resolution?

udenizdemirbilek, 2021-10-29:
Hello and thank you for this program that lets me use my 8 year old laptop!
I also encountered problems with the Accept button,
all my profiles are stuck to graphics card saving profile and I can't change the settings!
I'm attaching a video and my report file,
Also the following appears on the terminal when I click on the accept button

Lastly, I see you use Python for the program and was wondering if there's anything I can do to help, (I'm a novice programmer)

Take care!

[image]
[report_29-10-21_22:24.txt](https://github.com/slimbook/slimbookbattery/files/7444458/report_29-10-21_22.24.txt)

https://user-images.githubusercontent.com/81304552/139492079-cb5a0680-e580-44e9-b982-a46bbb49bf3b.mp4

slimbook, 2021-11-02:
Hi, thanks for reporting, your issue is related to [#21 ](https://github.com/slimbook/slimbookbattery/issues/21), the testing version, which will be released soon,  should fix it. Of course you can take a look at the code and [to do list](https://github.com/slimbook/slimbookbattery/projects/1#card-71584259) ;)

slimbook, 2021-11-15:
New version released

closed 2021-11-15 by slimbook