Slimbook-Team/slimbookbattery #48: Github actions

ehooo, 2021-11-04:
Hello,

I think may be is good idea to add a workflows on the project in order to allow pep8 checks on the PR and maybe also on the release could be added to build the deb, or other packages.

What do you think about?

I see that @MarSlimbook has remove .flake8 and a lot common files on .gitignore
There is any reason why?

Kind regards

ehooo, 2021-11-04:
https://github.com/ehooo/slimbookbattery/tree/pep8 this is a option if you like

slimbook, 2021-11-05:
Yes, I thought it would be a good idea to leave clean source code in the main branch, and work in the development branch (which has .flake8), and also creating a branch with package files, as purposed [here](https://github.com/slimbook/slimbookbattery/discussions/47#discussioncomment-1588098)

closed 2021-11-08 by slimbook