Slimbook-Team/slimbookbattery #55: Can't run application

MarSlimbook, 2021-11-15:
I have the same problem on Ubuntu 20.04!

_Originally posted by @Aminmalek in https://github.com/slimbook/slimbookbattery/issues/21#issuecomment-963475331_

MarSlimbook, 2021-11-15:
@Aminmalek could you share the new error log here?

bilalbayasut, 2021-11-19:
when running on terminal , it said 
```
ModuleNotFoundError: No module named 'gi'
```

MarSlimbook, 2021-11-22:
Hi @bilalbayasut try installing the dependency 'libgirepository-1.0-1', it is marked as a dependency so, did you installed it from the repository using apt? In that case, which distro are you using?

Aminmalek, 2021-11-22:
2021-11-22 19:23:30,330 - 665 - INFO - Intel detected
2021-11-22 19:23:30,338 - 690 - ERROR - TDP Controller not installed
in this new version(4.0.3) I can save the setting but nothing happens even brightness level doesn't change.

slimbook, 2021-11-23:
@Aminmalek, Is that the only output you get from the terminal? Could you generate a report (in the info window) and send it to us. The error of 'TDP controller not installed' does not stop the app.

bilalbayasut, 2021-11-23:
> Hi @bilalbayasut try installing the dependency 'libgirepository-1.0-1', it is marked as a dependency so, did you installed it from the repository using apt? In that case, which distro are you using?

xubuntu, I gave up, ended up installing ubuntu and it worked fine. thanks for your reply

dkocich, 2022-01-28:
I have the same issue, start in the terminal shows only an import problem even though I installed the APT package. I guess it is caused by multiple versions of python3 in the system.

❯ slimbookbattery
Traceback (most recent call last):
  File "/usr/share/slimbookbattery/src/slimbookbatterypreferences.py", line 21, in <module>
    import gi

slimbook, 2022-06-20:
Can't repro

dkocich, 2022-06-20:
@slimbook I tried again with beta
[image] - still the same issue appears in the terminal

[image]

i see pytgon-gi in my system packages
[image]

so I checked that my python is from linuxbrew - not using system packages...
[image]

slimbook, 2022-06-20:
Hi, so you are using Homebrew.
Check [Homebrew-and-Python](https://docs.brew.sh/Homebrew-and-Python)
Specially check if you did this:

> Follow the Next steps instructions to add Homebrew to your PATH and to your bash shell profile script, either ~/.profile on Debian/Ubuntu or ~/.bash_profile on CentOS/Fedora/Red Hat.

```
test -d ~/.linuxbrew && eval "$(~/.linuxbrew/bin/brew shellenv)"
test -d /home/linuxbrew/.linuxbrew && eval "$(/home/linuxbrew/.linuxbrew/bin/brew shellenv)"
test -r ~/.bash_profile && echo "eval \"\$($(brew --prefix)/bin/brew shellenv)\"" >> ~/.bash_profile
echo "eval \"\$($(brew --prefix)/bin/brew shellenv)\"" >> ~/.profile

```
This [StackOverlfow Issue](https://stackoverflow.com/questions/63246697/importing-modules-installed-via-homebrew-into-python) might also be helpful.
I'll try to take a closer look when I have some time.

closed 2023-07-29 by MarSlimbook