Slimbook-Team/slimbookbattery #56: Taskbar icon way too big

logileifs, 2021-11-20:
This is how the taskbar icon looks after the latest release (4.0.3)
[image]

I'm using 18.04.1-Ubuntu Budgie.
One thing I want to add as well, the old UI looked so much better, this new style is just way too much.

MarSlimbook, 2021-11-22:
Seems that this desktop indicator panel does not resize images, we will update as soon as we find a solution. Seems that converting to svg could work.

KavinAVS, 2021-12-12:
Yes, the off button is the same size.

[image]

slimbook, 2022-01-24:
Referenced to #62

closed 2022-01-24 by slimbook