Slimbook-Team/slimbookbattery #57: !!! HELP !!! GUI NOT SHOWING FULLY ON ELEMENTARYOS 5

amethylghostling, 2021-12-10:
[image]

MarSlimbook, 2021-12-13:
Hi, can you please check if configuration file '/home/<user>/.config/slimbookbattery/slibookbattery.conf' exists? (), then could you try restoring default values?

amethylghostling, 2021-12-16:
2021-12-16 14:11:42,380 - <module>:166 - INFO - Config check executed as root
2021-12-16 14:11:42,380 - main:42 - INFO - Configuration folder (/root/.config/slimbookbattery) foun
2021-12-16 14:11:42,380 - set_ownership:52 - DEBUG - Folder /root/.config/slimbookbattery
User uid=0
Folder uid=0
2021-12-16 14:11:42,380 - set_ownership:52 - DEBUG - Folder /usr/share/slimbookbattery/src/updates
User uid=0
Folder uid=0
2021-12-16 14:11:42,380 - check_config_file:68 - INFO - Checking Slimbook Battery Configuration
2021-12-16 14:11:42,382 - check_config_file:78 - INFO - Checking section: CONFIGURATION ...
2021-12-16 14:11:42,382 - check_config_file:78 - INFO - Checking section: SETTINGS ...
2021-12-16 14:11:42,382 - check_config_file:78 - INFO - Checking section: TDP ...
2021-12-16 14:11:42,382 - check_config_file:98 - INFO - Incidences not found.
2021-12-16 14:11:42,382 - check_tlp_files:109 - INFO - Checking Slimbook Battery's TLP Configuration
2021-12-16 14:11:42,382 - check_tlp_files:127 - INFO - Checking ahorrodeenergia
2021-12-16 14:11:42,386 - check_tlp_files:127 - INFO - Checking equilibrado
2021-12-16 14:11:42,391 - check_tlp_files:127 - INFO - Checking maximorendimiento
2021-12-16 14:11:42,395 - check_tlp_files:155 - INFO - Incidences not found.

**Results to python3 /usr/share/slimbookbattery/src/check_config.py  **

MarSlimbook, 2021-12-16:
Thank you, it's using a bad configuration route, I will try to reproduce the problem.

pranav2812, 2022-02-20:
Hey, I encountered the same issue on Ubuntu 18.04.6 LTS after installing slimbookbattery through ppa. 

This was indeed a path configuration error and was solved by changing path of tlp config file in **line 813** of  `/usr/share/slimbookbattery/src/slimbookbatterypreferences.py`  from `/etc/tlp.conf` to `/etc/default/tlp`.

It seems that the configuration file slimbookbattery is looking for resides at the later address (in my system atleast).

MarSlimbook, 2022-02-21:
@pranav2812 which version of TLP have you got? That directory changed since version 3.1. Have you tested if changes are being applied? I don't think changing that only line fixes the problem.

pranav2812, 2022-02-21:
My system has TLP 1.1 installed currently which I believe got installed when I installed slimbookbattery through the provided PPA for bionic. I'm not sure if the OP faces same issue as I did, (symptom seems to be same), but the path change solved the issue for me. After that, the program seems to be working as intended overall, I'm able to access the GUI and changes apply and persist.