Slimbook-Team/slimbookbattery #59: Incorrect usage of AppIndicator icons

p-himik, 2021-12-16:
**Describe the bug**
SlimBookBattery doesn't follow the correct usage of AppIndicator by using file paths as icon names, whereas proper icon names have to be provided. It leads to the indicator icon not being present or being just a placeholder in some circumstances, e.g. when using https://git.sr.ht/~steef/snixembed

The relevant code where such file paths are created: https://github.com/slimbook/slimbookbattery/blob/251a084eb67c5f031d1192c93871eb9c0f015497/src/slimbookbatteryindicator.py#L56-L58

Note that it might work with other systems because they might contain hacks to circumvent the root cause: https://github.com/ubuntu/gnome-shell-extension-appindicator/blob/ca543e59f939c2d0a7d81634418287f459d08c87/appIndicator.js#L559

Relevant places that mention that an icon name should be used and not a file path:
- https://gjs-docs.gnome.org/appindicator301~0.1_api/appindicator3.indicator#method-set_icon_full
- https://www.freedesktop.org/wiki/Specifications/StatusNotifierItem/StatusNotifierItem/#org.freedesktop.statusnotifieritem.iconname

Unfortunately, I'm not in the know to suggest the right fix here. Perhaps the source code of https://github.com/blueman-project/blueman could be consulted given that it's also written in Python and it seems to work just fine.

**Desktop (please complete the following information):**
 - OS: Ubuntu
 - Version: 21.10
 - Desktop Environment: KDE + Awesome WM

slimbook, 2022-01-24:
You were right, I modified it to use icon names, however, I'm not sure if it is compatible with https://git.sr.ht/~steef/snixembed
https://github.com/slimbook/slimbookbattery/blob/5a6d1e4edc987275ea3f328ecd467d278c943fdf/src/slimbookbatteryindicator.py#L53-L58
The .deb file can be found [here](https://github.com/slimbook/slimbookbattery/releases).

p-himik, 2022-02-06:
I think it works, thanks!

closed 2022-02-06 by p-himik