Slimbook-Team/slimbookbattery #60: sudocommands.py autostart crashes on Ubuntu 21.10

smiddy, 2021-12-24:
**Describe the bug**
Start Ubuntu 21.10 with installed (and autostarted) Slimbook battery

**To Reproduce**
Steps to reproduce the behavior:
1. Install slimbook
2. Login to your account
3. Ubuntu notifies about a crash (every time)
4. App works nevertheless

**Desktop (please complete the following information):**
 - OS: Xubuntu 64 bit
 - Version: 21.10
 - Desktop Environment: XFCE 4.16

**Additional context**
```
ProblemType: Crash
Date: Fri Dec 24 14:06:35 2021
ExecutablePath: /usr/share/slimbookbattery/src/sudocommands.py
ExecutableTimestamp: 1633429444
InterpreterPath: /usr/bin/python3.9
ProcCmdline: python3 /usr/share/slimbookbattery/src/sudocommands.py autostart
ProcCwd: /root
ProcEnviron:
 SHELL=/bin/sh
 LC_ADDRESS=de_CH.UTF-8
 LC_NAME=de_CH.UTF-8
 LC_MONETARY=de_CH.UTF-8
 LC_PAPER=de_CH.UTF-8
 LANG=en_US.UTF-8
 LC_IDENTIFICATION=de_CH.UTF-8
 LC_TELEPHONE=de_CH.UTF-8
 LC_MEASUREMENT=de_CH.UTF-8
 LC_TIME=de_CH.UTF-8
 PATH=(custom, no user)
 LC_NUMERIC=de_CH.UTF-8
ProcMaps:
 00400000-0041f000 r--p 00000000 fd:01 12058769                           /usr/bin/python3.9
 0041f000-00698000 r-xp 0001f000 fd:01 12058769                           /usr/bin/python3.9
 00698000-008e6000 r--p 00298000 fd:01 12058769                           /usr/bin/python3.9
 008e6000-008e7000 r--p 004e5000 fd:01 12058769                           /usr/bin/python3.9
 008e7000-0092e000 rw-p 004e6000 fd:01 12058769                           /usr/bin/python3.9
 0092e000-00952000 rw-p 00000000 00:00 0 
 0165b000-018e4000 rw-p 00000000 00:00 0                                  [heap]
 7f574c6ab000-7f574c87c000 rw-p 00000000 00:00 0 
 7f574c87c000-7f574c898000 r--p 00000000 fd:01 12059142                   /usr/lib/x86_64-linux-gnu/libssl.so.1.1
 7f574c898000-7f574c8ee000 r-xp 0001c000 fd:01 12059142                   /usr/lib/x86_64-linux-gnu/libssl.so.1.1
 7f574c8ee000-7f574c906000 r--p 00072000 fd:01 12059142                   /usr/lib/x86_64-linux-gnu/libssl.so.1.1
 7f574c906000-7f574c907000 ---p 0008a000 fd:01 12059142                   /usr/lib/x86_64-linux-gnu/libssl.so.1.1
 7f574c907000-7f574c910000 r--p 0008a000 fd:01 12059142                   /usr/lib/x86_64-linux-gnu/libssl.so.1.1
 7f574c910000-7f574c914000 rw-p 00093000 fd:01 12059142                   /usr/lib/x86_64-linux-gnu/libssl.so.1.1
 7f574c914000-7f574c923000 r--p 00000000 fd:01 12583443                   /usr/lib/python3.9/lib-dynload/_ssl.cpython-39-x86_64-linux-gnu.so
 7f574c923000-7f574c92e000 r-xp 0000f000 fd:01 12583443                   /usr/lib/python3.9/lib-dynload/_ssl.cpython-39-x86_64-linux-gnu.so
 7f574c92e000-7f574c93a000 r--p 0001a000 fd:01 12583443                   /usr/lib/python3.9/lib-dynload/_ssl.cpython-39-x86_64-linux-gnu.so
 7f574c93a000-7f574c93b000 r--p 00025000 fd:01 12583443                   /usr/lib/python3.9/lib-dynload/_ssl.cpython-39-x86_64-linux-gnu.so
 7f574c93b000-7f574c942000 rw-p 00026000 fd:01 12583443                   /usr/lib/python3.9/lib-dynload/_ssl.cpython-39-x86_64-linux-gnu.so
 7f574c942000-7f574ca43000 rw-p 00000000 00:00 0 
 7f574ca43000-7f574cabb000 r--p 00000000 fd:01 12059140                   /usr/lib/x86_64-linux-gnu/libcrypto.so.1.1
 7f574cabb000-7f574cc5e000 r-xp 00078000 fd:01 12059140                   /usr/lib/x86_64-linux-gnu/libcrypto.so.1.1
 7f574cc5e000-7f574ccec000 r--p 0021b000 fd:01 12059140                   /usr/lib/x86_64-linux-gnu/libcrypto.so.1.1
 7f574ccec000-7f574cd18000 r--p 002a8000 fd:01 12059140                   /usr/lib/x86_64-linux-gnu/libcrypto.so.1.1
 7f574cd18000-7f574cd1a000 rw-p 002d4000 fd:01 12059140                   /usr/lib/x86_64-linux-gnu/libcrypto.so.1.1
 7f574cd1a000-7f574cd1e000 rw-p 00000000 00:00 0 
 7f574cd26000-7f574cd28000 r--p 00000000 fd:01 12583389                   /usr/lib/python3.9/lib-dynload/_json.cpython-39-x86_64-linux-gnu.so
 7f574cd28000-7f574cd2f000 r-xp 00002000 fd:01 12583389                   /usr/lib/python3.9/lib-dynload/_json.cpython-39-x86_64-linux-gnu.so
 7f574cd2f000-7f574cd31000 r--p 00009000 fd:01 12583389                   /usr/lib/python3.9/lib-dynload/_json.cpython-39-x86_64-linux-gnu.so
 7f574cd31000-7f574cd32000 r--p 0000a000 fd:01 12583389                   /usr/lib/python3.9/lib-dynload/_json.cpython-39-x86_64-linux-gnu.so
 7f574cd32000-7f574cd33000 rw-p 0000b000 fd:01 12583389                   /usr/lib/python3.9/lib-dynload/_json.cpython-39-x86_64-linux-gnu.so
 7f574cd33000-7f574ce73000 rw-p 00000000 00:00 0 
[... 196 lines cut ...]
 NoNewPrivs:	0
 Seccomp:	0
 Seccomp_filters:	0
 Speculation_Store_Bypass:	thread vulnerable
 SpeculationIndirectBranch:	conditional enabled
 Cpus_allowed:	ff
 Cpus_allowed_list:	0-7
 Mems_allowed:	00000000,00000000,00000000,00000000,00000000,00000000,00000000,00000000,00000000,00000000,00000000,00000000,00000000,00000000,00000000,00000000,00000000,00000000,00000000,00000000,00000000,00000000,00000000,00000000,00000000,00000000,00000000,00000000,00000000,00000000,00000000,00000001
 Mems_allowed_list:	0
 voluntary_ctxt_switches:	170
 nonvoluntary_ctxt_switches:	145
PythonArgs: ['/usr/share/slimbookbattery/src/sudocommands.py', 'autostart']
Traceback:
 Traceback (most recent call last):
   File "/usr/share/slimbookbattery/src/sudocommands.py", line 618, in <module>
     main(sys.argv)
   File "/usr/share/slimbookbattery/src/sudocommands.py", line 131, in main
     battery_mode = config.get('CONFIGURATION', 'modo_actual')
   File "/usr/lib/python3.9/configparser.py", line 781, in get
     d = self._unify_values(section, vars)
   File "/usr/lib/python3.9/configparser.py", line 1152, in _unify_values
     raise NoSectionError(section) from None
 configparser.NoSectionError: No section: 'CONFIGURATION'
UserGroups: N/A
```

smiddy, 2022-01-16:
The same problem occurs on Ubuntu 20.04.3

slimbook, 2022-08-26:
#80

closed 2022-08-26 by slimbook