Slimbook-Team/slimbookbattery #61: The application Slimbook Battery has closed unexpectedly (Ubuntu 20.04.3)

baermathias, 2022-01-09:
**Describe the bug**
Crashes on startup with the following message:

[image]

**Desktop (please complete the following information):**
 - OS: Ubuntu 20.04.3

**Expected behavior (Optional)**
not to crash

ehooo, 2022-01-10:
Hello @baermathias 
Could you paste the fully logs?
It's looks like the user witch run the process don't have config file for the app.

baermathias, 2022-01-11:
It actually looks the same as this one https://github.com/slimbook/slimbookbattery/issues/60
The crash report appears in a GUI and there not much additional info, I also can't copy it, only screenshot.

MarSlimbook, 2022-01-12:
Hi, @baermathias, could you please check the content of /home/<your_user>/.config/slimbookbattery/slimbookbattery.conf?
Share the content of the file, then try to execute /usr/share/slimbookbattery/src/check_config.py (as sudo) and share the log of the execution again. 
It might have been an error in the creation of the file.

baermathias, 2022-01-14:
```python
# content of /home/my_user_name/.config/slimbookbattery/slimbookbattery.conf

[TDP]
tdpcontroller = slimbookintelcontroller
saving_tdpsync = 0

[CONFIGURATION]
application_on = 1
autostart = 1
icono = 1
max_battery_value = 90
min_battery_value = 15
max_battery_times = 2
time_between_warnings = 60
modo_actual = 2
show_time = 0
alerts = 0
hibernate_after_suspension = 0
style = original
plug_warn = 1
plugged = 2022-01-14

[SETTINGS]
limit_cpu_ahorro = 1
limit_cpu_equilibrado = 2
limit_cpu_maximorendimiento = 3
graphics_ahorro = 1
graphics_equilibrado = 1
ahorro_brightness = 30
equilibrado_brightness = 70
maxrendimiento_brightness = 90
ahorro_animations = 0
equilibrado_animations = 0
maxrendimiento_animations = 1
reset_config110720 = 1
graphics_maxrendimiento = 1
saving_brightness_switch = 1
balanced_brightness_switch = 1
power_brightness_switch = 1
```

```python
# log of check_config.py execution as sudo

2022-01-14 07:48:13,373 - <module>:166 - INFO - Config check executed as my_user_name
2022-01-14 07:48:13,373 - main:42 - INFO - Configuration folder (/home/my_user_name/.config/slimbookbattery) found!
2022-01-14 07:48:13,373 - set_ownership:52 - DEBUG - Folder /home/my_user_name/.config/slimbookbattery
User uid=1000
Folder uid=1000
2022-01-14 07:48:13,373 - set_ownership:52 - DEBUG - Folder /usr/share/slimbookbattery/src/updates
User uid=1000
Folder uid=1000
2022-01-14 07:48:13,373 - check_config_file:68 - INFO - Checking Slimbook Battery Configuration
2022-01-14 07:48:13,375 - check_config_file:78 - INFO - Checking section: CONFIGURATION ...
2022-01-14 07:48:13,376 - check_config_file:78 - INFO - Checking section: SETTINGS ...
2022-01-14 07:48:13,376 - check_config_file:78 - INFO - Checking section: TDP ...
2022-01-14 07:48:13,376 - check_config_file:98 - INFO - Incidences not found.
2022-01-14 07:48:13,376 - check_tlp_files:109 - INFO - Checking Slimbook Battery's TLP Configuration
2022-01-14 07:48:13,376 - check_tlp_files:127 - INFO - Checking ahorrodeenergia
2022-01-14 07:48:13,382 - check_tlp_files:127 - INFO - Checking equilibrado
2022-01-14 07:48:13,385 - check_tlp_files:127 - INFO - Checking maximorendimiento
2022-01-14 07:48:13,389 - check_tlp_files:155 - INFO - Incidences not found.

```

MarSlimbook, 2022-01-24:
It seems correct, so error might be at the configuration file path detection.

slimbook, 2022-01-24:
Moving to #60

closed 2022-01-24 by slimbook