Slimbook-Team/slimbookbattery #62: appIndicator/systray icons size for big screen resolutions

gentooza, 2022-01-24:
**Describe the bug**
Hi, I find the systray icons of slimbookbattery too big in the system tray, when using Mate in a slimbook executive 14 computer.
(2880x1800px)
A workaround for me is to manually overwrite: /usr/share/slimbookbattery/images/indicator/... png icons from 96x96px to 20x20px.
But I guess it should be adaptable. (vectorial?)

**To Reproduce**
Steps to reproduce the behavior:
I manually installed slimbookOs, and afterwards installed the ubuntu-mate-desktop flavour.

**Desktop (please complete the following information):**
 - OS: SlimbookOS
 - Version: 20.04
 - Desktop Environment: Mate

**Expected behavior (Optional)**
Icons in systray should be adaptable to screen resolution.

**Screenshots**
[image]
[image]

**Additional context**
Add any other context about the problem here.

slimbook, 2022-01-24:
Hi @gentooza, can you try this [testing version](https://github.com/slimbook/slimbookbattery/releases), it should fix this issue.

gentooza, 2022-01-26:
Hi!

Sorry, I didn't see #56

Yes, with the testing version icons are now 24x24 and fit perfectly.
When will be public in repo?

 thanks a lot!

closed 2022-04-11 by slimbook