Slimbook-Team/slimbookbattery #63: App Icon missing with update to 4.0.4

smiddy, 2022-01-30:
**Describe the bug**
Since the update to version 4.0.4, the icon symbols do not show anymore. The app is still running fine. And there is a blank place where it should show the current setting.
With the testing version [https://github.com/slimbook/slimbookbattery/releases/tag/Test](https://github.com/slimbook/slimbookbattery/releases/tag/Test) the problem is still there

**To Reproduce**
Steps to reproduce the behavior:
1. Install and look in the app tray
2. Click on '....'
3. Scroll down to '....'
4. See error

**Desktop (please complete the following information):**
 - OS: Xubuntu 21.10
 - Version: 4.0.4
 - Desktop Environment: Xfce

**Screenshots**
In the center of the image is the "invisible" icon tray.
[image]

TheProtonDev, 2022-03-03:
I'm having this same issue, except it is fixed when I start slimbook battery with sudo

I'm assuming this is a issue with permissions, where are the icons for slimbook stored?

MarSlimbook, 2022-03-04:
The icons are in /usr/share/slimbookbattery/images/indicator, @TheProtonDev do you also use Xubuntu?

TheProtonDev, 2022-03-04:
I use ArcoLinux XFCE, thanks, I'll try a few things and see if it fixes

smiddy, 2022-03-09:
I can confirm that executing

```sudo python3 /usr/share/slimbookbattery/src/slimbookbatteryindicator.py```

solves the problem.

smiddy, 2022-03-09:
Recursively giving read access of `usr/share/slimbookbattery/images` to the user does not solve the problem

smiddy, 2022-03-09:
@TheProtonDev @MarSlimbook, do you get the crashes as described in #60 ? This might be the source of the problem, if the program does get correct privileges.

TheProtonDev, 2022-03-09:
@smiddy 

I do not get any obvious errors, here's the logs outputted to the terminal when I start the program and hit apply to get the tray icon

```
⋊> ~ slimbookbattery                                                                                                                                              08:33:57
2022-03-09 09:45:03,024 - 1948 - DEBUG - Loading CSS - original
2022-03-09 09:45:03,057 - 1737 - DEBUG - Full window is displayed
2022-03-09 09:45:03,133 - 1858 - DEBUG - Width: 2560  Height: 1440
2022-03-09 09:45:03,168 - 733 - ERROR - TDP Controller not installed
2022-03-09 09:45:10,406 - 1977 - INFO - Closing window ...

2022-03-09 09:45:10,406 - 1998 - INFO - Rebooting slimbookbatteryindicator.py ...
2022-03-09 09:45:10,421 - 2021 - INFO - slimbookbatteryindicator.py was not running
2022-03-09 09:45:10,421 - 2024 - INFO - Launching process...
2022-03-09 09:45:10,424 - 2027 - INFO - Done

2022-03-09 09:45:10,424 - 2030 - INFO - 

2022-03-09 09:45:10,424 - 887 - INFO - Mode not setting TDP
2022-03-09 09:45:10,430 - 1498 - INFO - [ahorrodeenergia] Setting DEVICES_TO_DISABLE_ON_BAT_NOT_IN_USE saving to "bluetooth" --> Exit(0): 
2022-03-09 09:45:10,435 - 1498 - INFO - [ahorrodeenergia] Setting DEVICES_TO_DISABLE_ON_STARTUP saving to "" --> Exit(0): 
2022-03-09 09:45:10,441 - 1507 - INFO - [ahorrodeenergia] Setting DEVICES_TO_DISABLE_ON_LAN_CONNECT saving to "" --> Exit(0): 
2022-03-09 09:45:10,447 - 1507 - INFO - [ahorrodeenergia] Setting DEVICES_TO_ENABLE_ON_LAN_DISCONNECT saving to "" --> Exit(0): 
2022-03-09 09:45:10,452 - 1498 - INFO - [equilibrado] Setting DEVICES_TO_DISABLE_ON_BAT_NOT_IN_USE saving to "bluetooth" --> Exit(0): 
2022-03-09 09:45:10,458 - 1498 - INFO - [equilibrado] Setting DEVICES_TO_DISABLE_ON_STARTUP saving to "" --> Exit(0): 
2022-03-09 09:45:10,463 - 1507 - INFO - [equilibrado] Setting DEVICES_TO_DISABLE_ON_LAN_CONNECT saving to "" --> Exit(0): 
2022-03-09 09:45:10,469 - 1507 - INFO - [equilibrado] Setting DEVICES_TO_ENABLE_ON_LAN_DISCONNECT saving to "" --> Exit(0): 
2022-03-09 09:45:10,474 - 1498 - INFO - [maximorendimiento] Setting DEVICES_TO_DISABLE_ON_BAT_NOT_IN_USE saving to "" --> Exit(0): 
2022-03-09 09:45:10,480 - 1498 - INFO - [maximorendimiento] Setting DEVICES_TO_DISABLE_ON_STARTUP saving to "" --> Exit(0): 
2022-03-09 09:45:10,485 - 1507 - INFO - [maximorendimiento] Setting DEVICES_TO_DISABLE_ON_LAN_CONNECT saving to "" --> Exit(0): 
2022-03-09 09:45:10,494 - 1507 - INFO - [maximorendimiento] Setting DEVICES_TO_ENABLE_ON_LAN_DISCONNECT saving to "" --> Exit(0): 
2022-03-09 09:45:10,495 - 1974 - INFO - Not Gnome desktop "XFCE"
******************************************************************************
SlimbookBattery-Commandline, executed as: proton
Username: proton - Homedir: /home/proton
Arguments:  apply
Power saving mode

[BRIGTNESS SETTINGS]
/sys/class/backlight exists.
Setting Brightness (mode 1)  to 263 <--> 5 % ... | Exit: 0
cp: cannot create regular file '/var/spool/cron/crontabs/root': No such file or directory
sh: -c: line 2: unexpected EOF while looking for matching `''
sh: -c: line 4: syntax error: unexpected end of file
Crontab settings added

[TDP SETTINGS]
Battery Mode: 1
TDP Sync not active

[COPY TDP CUSTOM SETTINGS]
2022-03-09 09:45:10,624 - check_plug:263 - INFO - Time since last time disconnection: 5 days
File copied succesfully!
Already set in /etc/tlp.conf
TLP is enabled
******************************************************************************
SlimbookBattery-Commandline, executed as: proton
Username: proton - Homedir: /home/proton
Arguments:  autostart

[BRIGTNESS SETTINGS]
/sys/class/backlight exists.
Setting Brightness (mode 1)  to 263 <--> 5 % ... | Exit: 0
cp: cannot create regular file '/var/spool/cron/crontabs/root': No such file or directory
sh: -c: line 2: unexpected EOF while looking for matching `''
sh: -c: line 4: syntax error: unexpected end of file
Crontab settings added
******************************************************************************
2022-03-09 09:45:10,728 - __init__:114 - DEBUG - Brightness set
/usr/share/slimbookbattery/src/slimbookbatteryindicator.py:178: DeprecationWarning: Gtk.ImageMenuItem.set_image is deprecated
  item.set_image(icon)
/usr/share/slimbookbattery/src/slimbookbatteryindicator.py:179: DeprecationWarning: Gtk.ImageMenuItem.set_always_show_image is deprecated
  item.set_always_show_image(True)

[MODE SETTINGS]
Setting graphics frequency ...
Intel graphics power profile: low
Frequency not changed
```

slimbook, 2022-03-09:
@smiddy I think those are different issues, can you drop down the menu? (Even if you don't see the icon) might have something to do with the Xface desktop environment.

@TheProtonDev are brightness settings working for you?

smiddy, 2022-03-09:
Except for the tray icon, the manager works

STrRedWolf, 2022-03-25:
Throwing my two cents in here.  Same issue -- XUbuntu (XFCE) and the icon doesn't show... but click on the blank and the menu pops up, to where I can work the system or open the advanced window.

slimbook, 2022-03-28:
Hi, here you a have a [beta version](https://github.com/slimbook/slimbookbattery/releases), which has been updated recently. Try it please, thank you.

STrRedWolf, 2022-03-28:
The icon shows on the 4.0.5 beta.

smiddy, 2022-03-31:
I can confirm that the issue is resolvedwith the beta version.

closed 2022-04-11 by slimbook