Slimbook-Team/slimbookbattery #67: Slimbook is not saving battery.

Rank-N-Tensor, 2022-03-06:
**DESCRIPTION**
As the aforementioned title suggests, slimbook is not affecting battery saving. the time remaining is still the save as without slimbook ( about 1.5-2 hours with full charge)

**Evidence**
Steps to reproduce the behavior:
1. open terminal `Ctrl+Alt+T`
2. type `slimbookbattery`
3. select "energy Saving"
4. Restart System

**Desktop**
 - OS:  Ubuntu 20.04
 - Kernel Version: 5.13.0-30-generic
 - Desktop Environment: GNOME

**Expected behavior**
earlier, it would show ~10hrs with Energy Saving Mode

**Screenshots**
[image]

**Additional context**
TLP information:
```adi@Memory:~$ sudo tlp-stat -s
--- TLP 1.3.1 --------------------------------------------

+++ System Info
System         = ASUSTeK COMPUTER INC. 1.0        ROG Zephyrus G14 GA401IV_GA401IV
BIOS           = GA401IV.219
Release        = Ubuntu 20.04.4 LTS
Kernel         = 5.13.0-30-generic #33~20.04.1-Ubuntu SMP Mon Feb 7 14:25:10 UTC 2022 x86_64
/proc/cmdline  = BOOT_IMAGE=/boot/vmlinuz-5.13.0-30-generic root=UUID=1f12a647-18d4-4cc4-bb4a-514898c99010 ro quiet splash vt.handoff=7
Init system    = systemd v245 (245.4-4ubuntu3.15)
Boot mode      = UEFI

+++ TLP Status
State          = enabled
RDW state      = enabled
Last run       = 03:34:02 PM IST,    915 sec(s) ago
Mode           = battery
Power source   = battery
```
slimbook terminal output

```adi@Memory:~$ slimbookbattery
2022-03-06 15:50:01,553 - 1948 - DEBUG - Loading CSS - original
2022-03-06 15:50:01,580 - 1737 - DEBUG - Full window is displayed
2022-03-06 15:50:01,615 - 1858 - DEBUG - Width: 2560  Height: 1440
2022-03-06 15:50:01,678 - 733 - ERROR - TDP Controller not installed
2022-03-06 15:50:14,229 - 1977 - INFO - Closing window ...

2022-03-06 15:50:14,230 - 1998 - INFO - Rebooting slimbookbatteryindicator.py ...
2022-03-06 15:50:14,268 - 2009 - INFO - Killing process 2856 Exit: 0
2022-03-06 15:50:14,269 - 2013 - INFO - Launching process...
2022-03-06 15:50:14,271 - 2016 - INFO - Done
2022-03-06 15:50:14,271 - 2030 - INFO - 

2022-03-06 15:50:14,271 - 887 - INFO - Mode not setting TDP
2022-03-06 15:50:14,283 - 1498 - INFO - [ahorrodeenergia] Setting DEVICES_TO_DISABLE_ON_BAT_NOT_IN_USE saving to "bluetooth" --> Exit(0): 
2022-03-06 15:50:14,296 - 1498 - INFO - [ahorrodeenergia] Setting DEVICES_TO_DISABLE_ON_STARTUP saving to "" --> Exit(0): 
2022-03-06 15:50:14,310 - 1507 - INFO - [ahorrodeenergia] Setting DEVICES_TO_DISABLE_ON_LAN_CONNECT saving to "" --> Exit(0): 
2022-03-06 15:50:14,322 - 1507 - INFO - [ahorrodeenergia] Setting DEVICES_TO_ENABLE_ON_LAN_DISCONNECT saving to "" --> Exit(0): 
2022-03-06 15:50:14,334 - 1498 - INFO - [equilibrado] Setting DEVICES_TO_DISABLE_ON_BAT_NOT_IN_USE saving to "bluetooth" --> Exit(0): 
2022-03-06 15:50:14,346 - 1498 - INFO - [equilibrado] Setting DEVICES_TO_DISABLE_ON_STARTUP saving to "" --> Exit(0): 
2022-03-06 15:50:14,359 - 1507 - INFO - [equilibrado] Setting DEVICES_TO_DISABLE_ON_LAN_CONNECT saving to "" --> Exit(0): 
2022-03-06 15:50:14,371 - 1507 - INFO - [equilibrado] Setting DEVICES_TO_ENABLE_ON_LAN_DISCONNECT saving to "" --> Exit(0): 
2022-03-06 15:50:14,383 - 1498 - INFO - [maximorendimiento] Setting DEVICES_TO_DISABLE_ON_BAT_NOT_IN_USE saving to "" --> Exit(0): 
2022-03-06 15:50:14,394 - 1498 - INFO - [maximorendimiento] Setting DEVICES_TO_DISABLE_ON_STARTUP saving to "" --> Exit(0): 
2022-03-06 15:50:14,406 - 1507 - INFO - [maximorendimiento] Setting DEVICES_TO_DISABLE_ON_LAN_CONNECT saving to "" --> Exit(0): 
2022-03-06 15:50:14,419 - 1507 - INFO - [maximorendimiento] Setting DEVICES_TO_ENABLE_ON_LAN_DISCONNECT saving to "" --> Exit(0): 
2022-03-06 15:50:14,420 - 1954 - INFO - Setting mode 1 animations
2022-03-06 15:50:14,420 - 1967 - INFO - Animations Active
******************************************************************************
SlimbookBattery-Commandline, executed as: adi
Username: adi - Homedir: /home/adi
Arguments:  apply
Power saving mode

[BRIGTNESS SETTINGS]
/sys/class/backlight exists.
Setting Brightness (mode 1)  to 76 <--> 30 % ... | Exit: 0
Crontab settings added

[TDP SETTINGS]
Battery Mode: 1
TDP Sync not active

[COPY TDP CUSTOM SETTINGS]
File copied succesfully!
2022-03-06 15:50:14,626 - check_plug:263 - INFO - Time since last time disconnection: 1 days
Already set in /etc/tlp.conf
TLP is enabled
******************************************************************************
SlimbookBattery-Commandline, executed as: adi
Username: adi - Homedir: /home/adi
Arguments:  autostart

[BRIGTNESS SETTINGS]
/sys/class/backlight exists.
Setting Brightness (mode 1)  to 76 <--> 30 % ... | Exit: 0
Crontab settings added
******************************************************************************
2022-03-06 15:50:14,800 - __init__:114 - DEBUG - Brightness set
/usr/share/slimbookbattery/src/slimbookbatteryindicator.py:178: DeprecationWarning: Gtk.ImageMenuItem.set_image is deprecated
  item.set_image(icon)
/usr/share/slimbookbattery/src/slimbookbatteryindicator.py:179: DeprecationWarning: Gtk.ImageMenuItem.set_always_show_image is deprecated
  item.set_always_show_image(True)

[MODE SETTINGS]
Detected nvidia graphics profile: intel
Setting new profile ...
Graphics before --> intel // Graphics after --> intel
```
Additionally, as is evident from the screenshot. This unexpected behavior began when the quick select menu switched from "Energy Saving" +"Balanced" + "High Performance" to "Low"+"Medium"+"High".

Rank-N-Tensor, 2022-03-06:
Additionally, I found this warning in the TLP documentation.
[image]

i am running ubuntu 20.04.4 with the 5.13 kernel. It was only after the kernel update that this issue arose.

slimbook, 2022-03-07:
Thanks for sharing, have you tried the solution mentioned, or maybe a higher kernel version? We will also test it to compare.

Rank-N-Tensor, 2022-03-07:
Hi @slimbook, thanks for the prompt response. I did try the 5.11 kernel earlier, to no avail. My system (thankfully) does not freeze or reboot. although, earlier, the screen would become pixelated and i would be forced to shut down the laptop and restart it.
 
I havent implemented the solution as of now as im working some other sensitive projects and i am worried about unexpected behaviour. please advise(if you can) on how to work with DKMS and secure boot ( I am a beginner in this space).

slimbook, 2022-03-10:
I @Rank-N-Tensor, It's okey. Kernel 5.14 oem seems to work better with tlp and intel processors, lowering consumption, but we still need to do some research.
I'm afraid I can't help you much with DKMS or Secureboot.

Rank-N-Tensor, 2022-03-10:
understood, thank you for your time.

HamsterRICK, 2022-04-18:
I seem to be having the same issue, as described here. It doesn't matter what option I have select on menu the only thing it seems to be affecting is the backlight, and nothing else on the system, occasionally I get the battery meter to saying 6 Hours of battery life left, but it then goes back to the normal 2 hour left as if Slimbook Battery was not running.
I have tried running it on Xubuntu 20.04.4, PopOS V21.10 and beta version of Ubuntu 22.04, but nothing seem to work.