Slimbook-Team/slimbookbattery #70: Not enough values to unpack on Fedora 36

mpoder, 2022-04-09:
When running the Slimbook Battery application, we get a weird window with a width of 0 and a height of 0. If you expand the window to a more reasonable size, there is not anything else to see.

```
user@fedora ~> slimbookbattery
2022-04-10 01:23:39,987 - 1948 - DEBUG - Loading CSS - original
2022-04-10 01:23:39,998 - 1690 - ERROR - Unexpected error
Traceback (most recent call last):
  File "/usr/share/slimbookbattery/src/slimbookbatterypreferences.py", line 1688, in __init__
    self.set_ui()
  File "/usr/share/slimbookbattery/src/slimbookbatterypreferences.py", line 1735, in set_ui
    display_width, display_height = utils.get_display_resolution()
ValueError: not enough values to unpack (expected 2, got 1)
```

This happens when using Wayland - I have not yet tested this on X11.
Operating system is Fedora 36 - which - at the time of creating this bug report - is in beta.

ehooo, 2022-04-10:
Hello @mpoder 
Do you have xdpyinfo installed?

mpoder, 2022-04-12:
Hello @ehooo 
I... I did not have xdpyinfo installed, that was the reason it did not show up. Thanks!

Your suggestion did get the GUI working, but I was a little sad to see that I only had the General tab and nothing else really worked (including the app indicator, because it is built onto Ayatana, which is not really used much in Fedora as it appears). So currently the application does not seem very functional on Fedora.

Are you perhaps aware of any other dependencies that must be installed that the deploy script has not installed yet or does the application require more work to be compatible with Fedora?

slimbook, 2022-04-12:
Hello @mpoder,  I've made some changes in order to improve compatibility with Fedora, [here](https://github.com/slimbook/slimbookbattery/releases/tag/Fedora) there is a way to test it in fedora. There are some dependencies listed in the description, it would be great if you could check this out.

mpoder, 2022-04-14:
Hello @slimbook! Thank you for the Fedora version. This helped a lot + the additional dependencies came in handy as well. I actually seem to be getting the same error as I got before. 

The error is in the file `slimbookbattery/src/slimbookbatterypreferences.py` line 1354:
```
button = self.content['governor']
```

I also think I figured out the problem. My Slimbook on Fedora is using the (new) `amd-pstate` governor, which it seems the battery application does not have support for. Once I added code to the file above that respects the `amd-pstate` governor, then it ran fine.

I created a pull request that improved my situation here: https://github.com/slimbook/slimbookbattery/pull/72

closed 2022-06-20 by slimbook