Slimbook-Team/slimbookbattery #73: " you have not installed Linux-tools for your kernel"

Nationalgeographic, 2022-05-03:
**Describe the bug**
after updating Slimbookbattery from the AUR (I'm using Manjaro) I get a prompt, even tho I got the demanded "linux-tools-meta" package, see point 4. to reproduce. If I cklick on "install" nothing happens, like opening pamac/pacman package manager, or someting else.
 Slimbookbattery-gui also just show me the "general" tab.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to gui icon tray, 
2.  Right click and go for advanced mode 
4. See error: "You have not installed linux-tools for your kernel version, which is recommended. If you want to install it, click the 'Install' button below, otherwise, you can close this window, and everything will remain the same."

**Desktop (please complete the following information):**
 - OS:  Manjaro 
 - Version: Manjaro Linux x86_64  / Kernel 5.17.1.-3
 - Desktop Environment: KDE

**Expected behavior (Optional)**
Firstly no error prompt, secondly something should happen when clicked on "install".

**Screenshots**
[image]

Kind regards

slimbook, 2022-06-20:
Hi @Nationalgeographic, this is not fully implemented for arch yet, you can select the  'dont' show again checkbox' and Cancel button, It's just a recommendation, the app should work as always.
There is also an error caused by the configuration, try restoring default values.

slimbook, 2022-08-26:
Removed window in version 3.0.8

closed 2022-08-26 by slimbook