Slimbook-Team/slimbookbattery #76: Two batteries (Thinkpad T580)

wujastyk, 2022-06-27:
**Is your feature request related to a problem? Please describe.**

The Thinkpad T580 has two batteries: one internal and one removable (the usual kind).  As far as I can see, slimbookbattery only addresses the internal battery.  

**Describe the solution you'd like**

Slimbookbattery to recognize multiple batteries.

Thanks!

ehooo, 2022-08-25:
Hello @wujastyk 

This project use as backgroud [TLP](https://github.com/linrunner/TLP)

I think you need install `tp-smapi-dkms` and `acpi-call-dkms` as is describe in [the arch linux doc](https://wiki.archlinux.org/title/Talk:TLP)

vrossum, 2023-04-24:
I confirm this on my T480.
tlp is reporting it correctly (see below), but slimbookbattery is not.

tlp-stat -b
--- TLP 1.5.0 --------------------------------------------

+++ Battery Care
Plugin: thinkpad
Supported features: charge thresholds, recalibration
Driver usage:
* natacpi (thinkpad_acpi) = active (charge thresholds, recalibration)
Parameter value ranges:
* START_CHARGE_THRESH_BAT0/1:  0(off)..96(default)..99
* STOP_CHARGE_THRESH_BAT0/1:   1..100(default)

+++ ThinkPad Battery Status: BAT0 (Main / Internal)
/sys/class/power_supply/BAT0/manufacturer                   = SMP
/sys/class/power_supply/BAT0/model_name                     = 01AV421
/sys/class/power_supply/BAT0/cycle_count                    =    953
/sys/class/power_supply/BAT0/energy_full_design             =  24000 [mWh]
/sys/class/power_supply/BAT0/energy_full                    =  20070 [mWh]
/sys/class/power_supply/BAT0/energy_now                     =  15210 [mWh]
/sys/class/power_supply/BAT0/power_now                      =  15749 [mW]
/sys/class/power_supply/BAT0/status                         = Charging

/sys/class/power_supply/BAT0/charge_control_start_threshold =      0 [%]
/sys/class/power_supply/BAT0/charge_control_end_threshold   =    100 [%]
/sys/class/power_supply/BAT0/charge_behaviour               = [auto] inhibit-charge force-discharge

Charge                                                      =   75.8 [%]
Capacity                                                    =   83.6 [%]

+++ ThinkPad Battery Status: BAT1 (Ultrabay / Slice / Replaceable)
/sys/class/power_supply/BAT1/manufacturer                   = Celxpert
/sys/class/power_supply/BAT1/model_name                     = 01AV424
/sys/class/power_supply/BAT1/cycle_count                    =    420
/sys/class/power_supply/BAT1/energy_full_design             =  24050 [mWh]
/sys/class/power_supply/BAT1/energy_full                    =  19370 [mWh]
/sys/class/power_supply/BAT1/energy_now                     =  17340 [mWh]
/sys/class/power_supply/BAT1/power_now                      =      0 [mW]
/sys/class/power_supply/BAT1/status                         = Not charging

/sys/class/power_supply/BAT1/charge_control_start_threshold =      0 [%]
/sys/class/power_supply/BAT1/charge_control_end_threshold   =    100 [%]
/sys/class/power_supply/BAT1/charge_behaviour               = [auto] inhibit-charge force-discharge

Charge                                                      =   89.5 [%]
Capacity                                                    =   80.5 [%]

+++ Charge total                                            =   82.5 [%]

porteusconf, 2024-02-16:
Work-around... you can use tlp-UI control charge-levels of 2 batteries in lenovo X240 laptop, bat0 and bat1,
[image]
 as shown in screenshot.