Slimbook-Team/slimbookbattery #77: Slimbook breaks grub menu after installation (Linux Mint)

geraldhoarau14, 2022-08-21:
**Describe the bug**
The slimbook battery installation breaks my grub boot menu. It replaces my linux mint with ubuntu from grub menu. I have tried to reinstall grub, but nothing works.

**To Reproduce**
Steps to reproduce the behavior:
1. Install LInux MInt
2. Login into LInux MInt
3. Install slimbook battery 
4. restart the computer
5. LInux mint has been replaced by Ubuntu (broken)

**Desktop (please complete the following information):**
 - OS: Linux Mint
 - Version: 21 (vanessa)
 - Desktop Environment: CInnamon

**Expected behavior (Optional)**
LInux MInt should not be replaced by Ubuntu from Grub. I should be able to login into LInux MInt from grub.

slimbook, 2022-08-22:
This is new, have you installed linux-tools after slimbookbattery installation? It seems more like a linux-mint issue.
About the grub, you can try boot-repair:
```
sudo add-apt-repository ppa:yannubuntu/boot-repair && sudo apt update
sudo apt install -y boot-repair && boot-repair
```
Take a look at this:
https://forums.linuxmint.com/viewtopic.php?p=1853804#p1853804

geraldhoarau14, 2022-08-23:
Thank you very much for your reply. I haven't installed the linux-tools. I will try to install it before the slimbook battery installation next time.

I'm not sure which side is responsible for the issue, but it happened to me twice after the slimbook battery installation. I was able to reboot the system and login without any problems before the installation. But Grub has become corrupted after the slimbook battery installation. 

I have tried your suggestion as well. but it didn't work in my case. I'm still a beginner with LInux, so I assume  that my skill is not good enough and the problem is probably coming from my side. I'm not sure if I have installed the application properly.

I will try to install the app again and let you know if the issue happens again.

geraldhoarau14, 2022-08-23:
I have reinstall slimbook battery, and it didn't break grub this time. This is what I make differently.

 - I have installed the linux-tools before the installation.
 - I have installed the app directly from the .deb package rather than installing from the package manager by adding the repository manually.

slimbook, 2022-08-24:
Thank you for the details,  however, I find no relation to slimbookbattery (since it has nothing to do with grub) so I'm closing this issue.
If someone finds a similar issue I will reopen it.

closed 2022-08-24 by slimbook