Slimbook-Team/slimbookbattery #78: configparser.NoOptionError: No option 'saving_tdpsync' in section: 'TDP'

ERROR-404-NULL-NOT-FOUND, 2022-08-22:
I'm on Arch Linux, installing the `slimbookbattery` package from the AUR.
Whenever I try to run the program, it shows the error:
```
Traceback (most recent call last):
  File "/usr/lib/python3.10/configparser.py", line 790, in get
    value = d[option]
  File "/usr/lib/python3.10/collections/__init__.py", line 986, in __getitem__
    return self.__missing__(key)            # support subclasses that define __missing__
  File "/usr/lib/python3.10/collections/__init__.py", line 978, in __missing__
    raise KeyError(key)
KeyError: 'saving_tdpsync'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/share/slimbookbattery/src/slimbookbatterypreferences.py", line 1700, in __init__
    self.set_ui()
  File "/usr/share/slimbookbattery/src/slimbookbatterypreferences.py", line 1887, in set_ui
    self.general_page_grid = GeneralGrid(self)
  File "/usr/share/slimbookbattery/src/slimbookbatterypreferences.py", line 663, in __init__
    super(GeneralGrid, self).__init__(parent, *args, **kwargs)
  File "/usr/share/slimbookbattery/src/slimbookbatterypreferences.py", line 113, in __init__
    self.complete_values()
  File "/usr/share/slimbookbattery/src/slimbookbatterypreferences.py", line 795, in complete_values
    button.set_active(config.getboolean('TDP', field))
  File "/usr/lib/python3.10/configparser.py", line 829, in getboolean
    return self._get_conv(section, option, self._convert_to_boolean,
  File "/usr/lib/python3.10/configparser.py", line 809, in _get_conv
    return self._get(section, conv, option, raw=raw, vars=vars,
  File "/usr/lib/python3.10/configparser.py", line 804, in _get
    return conv(self.get(section, option, **kwargs))
  File "/usr/lib/python3.10/configparser.py", line 793, in get
    raise NoOptionError(option, section)
configparser.NoOptionError: No option 'saving_tdpsync' in section: 'TDP'
```
And displays a window without any options other than `System Style`, `Cancel`, `Restore Default Values`, and `Accept`

slimbook, 2022-08-22:
Try restoring default values, that should fix configuration issues.

ERROR-404-NULL-NOT-FOUND, 2022-08-22:
Clicking `Restore Default Values` Just made the window shrink to a tiny, maybe 4x4 pixel back window, and when I killed and restarted it, absolutely nothing changed.