Slimbook-Team/slimbookbattery #79: Hello!

KValiullina07, 2022-08-23:
I got Kernel Panic after installing slimbook battery.

I started it in "maximum performance".

Can you help me please?[image]
[image]

slimbook, 2022-08-24:
Hello, did you install linux-tools from a window when you opened the app?

slimbook, 2022-09-01:
HI, we have received s a similar issue mentioning that it happens when high performance mode is set, and then reboot. Does this  match your situation? In that case it's possible that it has something to do with Nvidia settings