Slimbook-Team/slimbookbattery #88: Restore TLP config when turned off

siddhpant, 2022-11-06:
**Is your feature request related to a problem? Please describe.**
Turning off Slimbook Battery (SBAT henceforth) doesn't restore my previous TLP settings and neither does it turn it on. I expected SBAT to pass control to TLP when I turn SBAT off, rather than stop power management in the laptop altogether.

**Describe the solution you'd like**
Turn on TLP and restore old config when I turn off SBAT.

**Describe alternatives you've considered**
Manually restoring from backup and running `sudo tlp start`.

**Additional context**
I also had set some settings in TLP which SBAT maybe seems to override, like disabling radio devices like BT on startup. I think it should not do that, there is more power saving in them disabled on start, and since it is set by user, high performance should not enable it either (it doesn't depend on it anyways).

siddhpant, 2022-11-07:
> There is a setting on each mode that does this, so if the switch is off, when you click the accept button it will override that parameter with the selected setting. When you install the app default settings are shown, maybe it should use your custom settings if exist.

I only see the General tab on 4.0.4. I installed the latest from repo*, and now I can see the options.

I guess it's due to the below keyerror which was fixed:
```
Traceback (most recent call last):
  File "/usr/share/slimbookbattery/src/slimbookbatterypreferences.py", line 1688, in __init__
    self.set_ui()
  File "/usr/share/slimbookbattery/src/slimbookbatterypreferences.py", line 1871, in set_ui
    SettingsGrid(self, data.get('filename'))
  File "/usr/share/slimbookbattery/src/slimbookbatterypreferences.py", line 1172, in __init__
    super(SettingsGrid, self).__init__(parent, *args, **kwargs)
  File "/usr/share/slimbookbattery/src/slimbookbatterypreferences.py", line 108, in __init__
    self.complete_values()
  File "/usr/share/slimbookbattery/src/slimbookbatterypreferences.py", line 1344, in complete_values
    button = self.content['governor']
KeyError: 'governor'
```

\* There are some problems with deploy script, I will try to send a fix.

siddhpant, 2022-11-07:
Also, is there any way to set default profile (or settings to apply automatically when on AC)?

siddhpant, 2022-11-07:
> Do you mean this [default tlp settings](https://github.com/linrunner/TLP/blob/main/tlp.conf) which has all params commented? If not, there is a button that says "restore default values" which sets our default settings for each mode. You can see them at /usr/share/slimbookbattery/default/, the three files contain default setting for each mode.

I mean when I connect AC, can SBAT automatically switch settings / mode (like change to high performance), like how TLP automatically changes settings and governor when I connect charger?

> sorry for not having translated the names, ahorrodeenergia = low, equilibrado = balanced, and maximorendimiento = high.

np, and thanks!

slimbook, 2022-11-07:
> I mean when I connect AC, can SBAT automatically switch settings / mode (like change to high performance), like how TLP automatically changes settings and governor when I connect charger?

Now I understand. 
That's because there are params that only act when batt is connected, for example, the governor settings.
In this case, as we only set the ones which act when battery is used, so, once you connect the ac, governor will switch to performance , and stop using the bat settings (1st column).
You can check it with `sudo tlp stat -p`

siddhpant, 2022-11-07:
> once you connect the ac, governor will switch to performance , and stop using the bat settings (1st column). You can check it with `sudo tlp stat -p`

It doesn't change when I connect AC. I will use monitor mode of `auto-cpufreq` to show this (since it uses an update loop and shows battery status).

I set energy saving mode, and the output is as below which is expected (ignore suggestion):

[image]

Now when I connect battery, the output is as below:

[image]

So it didn't change.

If I change the mode to maximum performance, the output still remains the same, because the 1st column settings don't apply when on AC:

[image]

Here is the max performance setting for reference:

[image]

Also, it seems the "Disable animation" switch has negative logic. If I turn it on, then animations are enabled. Maybe it was intended to be yes/no, but then implemented as a switch.

I guess this should be in a new issue...