Slimbook-Team/slimbookbattery #90: Regarding changing settings on connection to AC

siddhpant, 2022-11-07:
_Originally posted by me in https://github.com/Slimbook-Team/slimbookbattery/issues/88#issuecomment-1305484243._

_Since it was too off-topic for that issue, hence created a new one._

---

> once you connect the ac, governor will switch to performance , and stop using the bat settings (1st column). You can check it with `sudo tlp stat -p`

It doesn't change when I connect AC. I will use monitor mode of `auto-cpufreq` to show this (since it uses an update loop and shows battery status).

I set energy saving mode, and the output is as below which is expected (ignore suggestion):

[image]

Now when I connect battery, the output is as below:

[image]

So it didn't change.

If I change the mode to maximum performance, the output still remains the same, because the 1st column settings don't apply when on AC:

[image]

Here is the max performance setting for reference:

[image]

Also, it seems the "Disable animation" switch has negative logic. If I turn it on, then animations are enabled. Maybe it was intended to be yes/no, but then implemented as a switch.

slimbook, 2022-11-07:
I see, in my tests does not happen, can you attach the file /etc/tlp.conf please?

siddhpant, 2022-11-07:
> I see, in my tests does not happen, can you attach the file /etc/tlp.conf please?

https://gist.github.com/siddhpant/86b13502b7787307baf526797b9297ff

slimbook, 2022-11-08:
Try using the restore default values button, accept changes, and check that CPU_SCALING_GOVERNOR_ON_AC parameter is not empty.

slimbook, 2022-11-10:
I've detected that this happens only on amd laptops (I was testing with an intel laptop), I'm working to fix this.

slimbook, 2022-11-10:
If you don't want to wait for the update you can update the default configuration locally as in the above commit, and restore default values.

siddhpant, 2022-11-10:
Hello, sorry for replying late.

> Try using the restore default values button, accept changes, and check that CPU_SCALING_GOVERNOR_ON_AC parameter is not empty.

I restored the default values, and checked. It is indeed empty.

```
$ grep CPU_SCALING_GOVERNOR_ON_AC /etc/tlp.conf
CPU_SCALING_GOVERNOR_ON_AC=
```

> I've detected that this happens only on low config mode, I'm working to fix this.

I'm not sure what this means. If you are saying it won't happen if I am on other modes like balanced and max perf, then that is incorrect since it happens in it too.

```
$ grep CPU_SCALING_GOVERNOR_ON_AC .config/slimbookbattery/default/*
.config/slimbookbattery/default/ahorrodeenergia:CPU_SCALING_GOVERNOR_ON_AC=
.config/slimbookbattery/default/equilibrado:CPU_SCALING_GOVERNOR_ON_AC=
.config/slimbookbattery/default/maximorendimiento:CPU_SCALING_GOVERNOR_ON_AC=
```

And yeah, I am on an AMD laptop.