Slimbook-Team/slimbookbattery #96: slimbookbattery wont start on debian

Am0rphous, 2023-01-17:
**Describe the bug**
When installing the software from either the git repo or from deb files, the software won't start.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to [this](https://software.opensuse.org//download.html?project=home%3ASlimbook%3ASlimbookBattery&package=slimbookbattery) link and download the deb file for debian
2. Install it with `sudo dpkg -i blabla.deb`
3. Open a new terminal window and run `slimbookbattery` or search and open it from the menu. It does not appear and there is no processes running named similar to slim: `ps aux|grep slim`. No error messages in the terminal either.

**Desktop (please complete the following information):**
 - OS: Kali GNU/Linux Rolling 
 - Version: 2022.4
 - Kernel: 6.0.0-kali6-amd64
 - Desktop Environment: GDM3

**Expected behavior (Optional)**
A tryicon is supposed to appear, but here is none.

**Additional context**
Installing it from the repo gives following output:
````
└─$ ./deploy.sh 

This script will install SlimbookBattery in your system (replacing previous
installation if exists).
Press ENTER to proceed, or CTRL-C to exit.

Checking and installing the following system dependencies (if needed)...
tlp tlp-rdw libnotify-bin cron gobject-introspection libayatana-appindicator3-1

...Done installing system dependencies!

Installing Python dependencies...
...Done installing Python dependencies!

Installing SlimbookBattery...

Removing previously installed resources...
[sudo] password for [REMOVED]: 
...Done removing previously installed resources!

Creating folder structure...
...Done creating folder structure!

Copying contents...
'bin/slimbookbattery' -> '/usr/bin/slimbookbattery'
'bin/slimbookbattery-pkexec' -> '/usr/bin/slimbookbattery-pkexec'
'bin/slimbookbattery' -> '/usr/share/slimbookbattery/bin/slimbookbattery'
'bin/slimbookbattery-pkexec' -> '/usr/share/slimbookbattery/bin/slimbookbattery-pkexec'
cp: cannot overwrite directory '/usr/share/slimbookbattery/changelog' with non-directory
'slimbookbattery.desktop' -> '/usr/share/applications/slimbookbattery.desktop'
'src/ac_power.py' -> '/usr/share/slimbookbattery/src/ac_power.py'
'src/check_config.py' -> '/usr/share/slimbookbattery/src/check_config.py'
'src/configuration' -> '/usr/share/slimbookbattery/src/configuration'
'src/configuration/custom' -> '/usr/share/slimbookbattery/src/configuration/custom'
'src/configuration/custom/ahorrodeenergia' -> '/usr/share/slimbookbattery/src/configuration/custom/ahorrodeenergia'
'src/configuration/custom/equilibrado' -> '/usr/share/slimbookbattery/src/configuration/custom/equilibrado'
'src/configuration/custom/maximorendimiento' -> '/usr/share/slimbookbattery/src/configuration/custom/maximorendimiento'
'src/configuration/default' -> '/usr/share/slimbookbattery/src/configuration/default'
'src/configuration/default/ahorrodeenergia' -> '/usr/share/slimbookbattery/src/configuration/default/ahorrodeenergia'
'src/configuration/default/equilibrado' -> '/usr/share/slimbookbattery/src/configuration/default/equilibrado'
'src/configuration/default/maximorendimiento' -> '/usr/share/slimbookbattery/src/configuration/default/maximorendimiento'
'src/configuration/default/tlp' -> '/usr/share/slimbookbattery/src/configuration/default/tlp'
'src/configuration/default/tlp/default-1.3.conf' -> '/usr/share/slimbookbattery/src/configuration/default/tlp/default-1.3.conf'
'src/configuration/default/tlp/default-1.5.conf' -> '/usr/share/slimbookbattery/src/configuration/default/tlp/default-1.5.conf'
'src/configuration/default/tlp/old_defaults' -> '/usr/share/slimbookbattery/src/configuration/default/tlp/old_defaults'
'src/configuration/default/tlp/old_defaults/ahorrodeenergia' -> '/usr/share/slimbookbattery/src/configuration/default/tlp/old_defaults/ahorrodeenergia'
'src/configuration/default/tlp/old_defaults/equilibrado' -> '/usr/share/slimbookbattery/src/configuration/default/tlp/old_defaults/equilibrado'
'src/configuration/default/tlp/old_defaults/maximorendimiento' -> '/usr/share/slimbookbattery/src/configuration/default/tlp/old_defaults/maximorendimiento'
'src/configuration/default/tlp/tlp-intrinsic-defaults.conf' -> '/usr/share/slimbookbattery/src/configuration/default/tlp/tlp-intrinsic-defaults.conf'
[... 118 lines cut ...]
  File "/usr/share/slimbookbattery/src/check_config.py", line 175, in <module>
    main()
  File "/usr/share/slimbookbattery/src/check_config.py", line 50, in main
    set_ownership(UPDATES_DIR)
  File "/usr/share/slimbookbattery/src/check_config.py", line 66, in set_ownership
    os.chown(dir_path, uid, gid)
PermissionError: [Errno 1] Operation not permitted: '/usr/share/slimbookbattery/src/updates'
chmod: changing permissions of '/usr/share/slimbookbattery/src/service/slimbookbatteryservice.sh': Operation not permitted
chmod: changing permissions of '/usr/share/slimbookbattery/src/updates/indicator_extension.sh': Operation not permitted
chmod: changing permissions of '/usr/share/slimbookbattery/src/ac_power.py': Operation not permitted
chmod: changing permissions of '/usr/share/slimbookbattery/src/check_config.py': Operation not permitted
chmod: changing permissions of '/usr/share/slimbookbattery/src/slimbookbatteryindicator.py': Operation not permitted
chmod: changing permissions of '/usr/share/slimbookbattery/src/slimbookbatterypreferences.py': Operation not permitted
chmod: changing permissions of '/usr/share/slimbookbattery/src/slimbookbattery-report.py': Operation not permitted
chmod: changing permissions of '/usr/share/slimbookbattery/src/splash.py': Operation not permitted
chmod: changing permissions of '/usr/share/slimbookbattery/src/sudocommands.py': Operation not permitted
chmod: changing permissions of '/usr/share/slimbookbattery/src/tdp_utils.py': Operation not permitted
chmod: changing permissions of '/usr/share/slimbookbattery/src/translations/comparer.py': Operation not permitted
chmod: changing permissions of '/usr/share/slimbookbattery/src/utils.py': Operation not permitted
Synchronizing state of tlp.service with SysV service script with /lib/systemd/systemd-sysv-install.
Executing: /lib/systemd/systemd-sysv-install enable tlp
...Done installing SlimbookBattery!
````

Any tip?

siddhpant, 2023-01-28:
Kali is not Debian. It is a different distro based on Debian. I don't encounter this on my Debian machine and can run Slimbook Battery successfully everyday.

siddhpant, 2023-01-28:
>  Installing it from the repo gives following output:

Please run it as root. It should ask. Will send a fix.

siddhpant, 2023-01-28:
#100 should fix installation.

closed 2023-02-02 by MarSlimbook