Slimbook-Team/slimbookbattery #104: Configuration tabs are gone

utrack, 2023-05-05:
**Describe the bug**
When I open the UI I don't see any tabs for configuration. I've seen them before - not sure where to even start debugging?
Arch Linux, installed from AUR.

It does not seem to be a style issue - I've toggled System style and tabs seem to be gone with that style as well.

**Desktop (please complete the following information):**
 - OS: Arch Linux
 - Version: rolling latest
 - Desktop Environment: KDE+i3
 - System: Lenovo T14s Gen3 AMD

**Expected behavior (Optional)**
I can see the config tabs.

**Screenshots**
[image]

**Additional context**
Logs:
```
» slimbookbattery                                                                                          2023-05-05 15:30:45
Cannot open log file /var/slimbookbattery.log, using /tmp/slimbookbattery.log

(slimbookbatterypreferences.py:64550): Gtk-WARNING **: 15:35:11.918: Theme parsing error: gtk.css:1649:16: '-gtk-icon-size' is not a valid property name

(slimbookbatterypreferences.py:64550): Gtk-WARNING **: 15:35:11.918: Theme parsing error: gtk.css:1652:16: '-gtk-icon-size' is not a valid property name
2023-05-05 15:35:12,190 - 727 - ERROR - TDP Controller not installed
2023-05-05 15:35:12,280 - 1692 - ERROR - Unexpected error
Traceback (most recent call last):
  File "/usr/share/slimbookbattery/src/slimbookbatterypreferences.py", line 1690, in __init__
    self.set_ui()
  File "/usr/share/slimbookbattery/src/slimbookbatterypreferences.py", line 1895, in set_ui
    SettingsGrid(self, data.get('filename'))
  File "/usr/share/slimbookbattery/src/slimbookbatterypreferences.py", line 1166, in __init__
    super(SettingsGrid, self).__init__(parent, *args, **kwargs)
  File "/usr/share/slimbookbattery/src/slimbookbatterypreferences.py", line 113, in __init__
    self.complete_values()
  File "/usr/share/slimbookbattery/src/slimbookbatterypreferences.py", line 1350, in complete_values
    button = self.content['governor']
             ~~~~~~~~~~~~^^^^^^^^^^^^
KeyError: 'governor'
Cannot open log file /var/slimbookbattery.log, using /tmp/slimbookbattery.log

(splash.py:64596): Gtk-WARNING **: 15:35:12.452: Theme parsing error: gtk.css:1649:16: '-gtk-icon-size' is not a valid property name

(splash.py:64596): Gtk-WARNING **: 15:35:12.453: Theme parsing error: gtk.css:1652:16: '-gtk-icon-size' is not a valid property name
Using  /etc/tlp.conf
2023-05-05 15:35:12,683 - u - <module>:73 - DEBUG - ******************************************************************************
2023-05-05 15:35:12,684 - u - <module>:75 - INFO - SlimbookBattery-Commandline, executed as: u
2023-05-05 15:35:12,684 - u - <module>:77 - DEBUG - Username: u - Homedir: /home/u
2023-05-05 15:35:12,685 - u - main:136 - DEBUG - Arguments:  restore
2023-05-05 15:35:12,685 - u - main:154 - INFO - Resetting modes conf
2023-05-05 15:35:12,686 - u - main:174 - INFO - Resetting Slimbook Battery's conf
2023-05-05 15:35:12,686 - u - main:177 - INFO - /usr/share/slimbookbattery/src/configuration/slimbookbattery.conf
2023-05-05 15:35:12,687 - u - main:216 - DEBUG - ******************************************************************************
<Gtk.Grid object at 0x7fc750d90d00 (GtkGrid at 0x5568dd6d9a00)>
2023-05-05 15:35:12,771 - 727 - ERROR - TDP Controller not installed
2023-05-05 15:35:12,858 - 1702 - ERROR - Unexpected error
Traceback (most recent call last):
  File "/usr/share/slimbookbattery/src/slimbookbatterypreferences.py", line 1690, in __init__
    self.set_ui()
  File "/usr/share/slimbookbattery/src/slimbookbatterypreferences.py", line 1895, in set_ui
    SettingsGrid(self, data.get('filename'))
  File "/usr/share/slimbookbattery/src/slimbookbatterypreferences.py", line 1166, in __init__
    super(SettingsGrid, self).__init__(parent, *args, **kwargs)
  File "/usr/share/slimbookbattery/src/slimbookbatterypreferences.py", line 113, in __init__
    self.complete_values()
  File "/usr/share/slimbookbattery/src/slimbookbatterypreferences.py", line 1350, in complete_values
    button = self.content['governor']
             ~~~~~~~~~~~~^^^^^^^^^^^^
KeyError: 'governor'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/share/slimbookbattery/src/slimbookbatterypreferences.py", line 1700, in __init__
    self.set_ui()
  File "/usr/share/slimbookbattery/src/slimbookbatterypreferences.py", line 1895, in set_ui
    SettingsGrid(self, data.get('filename'))
  File "/usr/share/slimbookbattery/src/slimbookbatterypreferences.py", line 1166, in __init__
    super(SettingsGrid, self).__init__(parent, *args, **kwargs)
  File "/usr/share/slimbookbattery/src/slimbookbatterypreferences.py", line 113, in __init__
    self.complete_values()
  File "/usr/share/slimbookbattery/src/slimbookbatterypreferences.py", line 1350, in complete_values
    button = self.content['governor']
             ~~~~~~~~~~~~^^^^^^^^^^^^
KeyError: 'governor'
Traceback (most recent call last):
  File "/usr/share/slimbookbattery/src/slimbookbatterypreferences.py", line 1953, in manage_events
    self.apply_conf()
  File "/usr/share/slimbookbattery/src/slimbookbatterypreferences.py", line 2012, in apply_conf
    self.low_page_grid.save_selection()
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'save_selection'

took 9s
~
»                                                                                                          2023-05-05 15:35:20
```

siddhpant, 2023-05-05:
Sometimes it may happen when it cannot detect governors. Happened with me once when the following file did not exist due to some signing issues on my part:

`/sys/devices/system/cpu/cpu0/cpufreq/scaling_available_governors`

Check if it lists all the governors.

siddhpant, 2023-06-24:
> Yep! I've added `amd_pstate_epp` to a couple of different switches in the slimbookbatterysettings.py and now the tabs are there.

Can you make a PR?

byteManiak, 2023-07-03:
I can reproduce this on an Arch Linux install with kernel 6.4.1. Adding 'amd-pstate-epp' to the various arrays/switches in the script does fix it.