Slimbook-Team/slimbookbattery #105: Support for Ubuntu Lunar

cHunter789, 2023-05-23:
Currently there is no repository for Ubuntu 23.04. Could you please add?

siddhpant, 2023-05-24:
You need to install it yourself. All repos are outdated.

cHunter789, 2023-05-26:
The problem is bigger. The installation script uses pip without a venv what is forbidden in new Ubuntu and Fedora and give me error:

> _error: externally-managed-environment_

I have to modify this part of deploy.sh script:
```
if [[ ! $(pip3 install -r requirements.txt) ]]; then
        print_warn_bold "ERROR: Could not install Python dependencies."
        exit 1
fi
```
To this one (will work only for Ubuntu):
```
if [[ ! $(apt-get install python-gi-dev python-configparser) ]]; then
        print_warn_bold "ERROR: Could not install Python dependencies."
        exit 1
fi
```
Other solution could be to use pipx as described here:
[](https://itsfoss.com/externally-managed-environment/)

closed 2023-05-26 by cHunter789