Slimbook-Team/slimbookbattery #110: Slimbook Battery is not starting even though no other instance of it is running

byteManiak, 2023-07-03:
**Description**
Slimbook Battery is not starting, it only says "Slimbook Battery is in execution". However, no other instance of it is running.

**Evidence**
Steps to reproduce the behavior:
1. Run `slimbookbattery` from terminal

**Desktop (please complete the following information):**
 - OS: Arch Linux
 - Version: latest Linux kernel as of this ticket (6.4.1)
 - Desktop Environment: Plasma Wayland

**Expected behavior**
The Slimbook Battery GUI shows

**Additional context**
It looks like there is a problem with the Python script that starts the GUI. In the snippet below, pgrep returns greater than 1 even though there is no `slimbookbatterypreferences` running:

`pgrep = subprocess.getoutput("pgrep -f slimbookbatterypreferences")
    num_proc_running = pgrep.split('\n')
    if len(num_proc_running) > 1:
        sys.stdout.write("Slimbook Battery is in execution\n")
    elif args.minimize:
        subprocess.run(['/usr/bin/env', 'python3', '/usr/share/slimbookbattery/src/slimbookbatteryindicator.py'])
    else:
        subprocess.run(['/usr/bin/env', 'python3', '/usr/share/slimbookbattery/src/slimbookbatterypreferences.py'])`

Output of `pgrep -f slimbookbatterypreferences` is blank:
`[byte@Lenny ~]$ pgrep -f slimbookbatterypreferences
[byte@Lenny ~]$`

brisvag, 2023-07-04:
Having the same issues; the code on `main` actually contains the fix, it's just unreleased. Can we get a release @slimbook ?

slimbook, 2023-07-05:
This is not tested yet, but I built a binary:
https://github.com/Slimbook-Team/slimbookbattery/releases/tag/v4.0.9beta
I'll test it tomorrow and upload it to launchpad if everything is okey

slarse, 2023-07-11:
@slimbook I have the exact same problem as @byteManiak using slimbookbattery v4.0.8. My setup is similar:

**Desktop (please complete the following information):**

- OS: Arch Linux
- Version: Linux kernel 6.4.2
- Desktop Environment: i3 with a cobbled together mess of various DE parts I like

v4.0.9 you linked above solves the problem for me.