Slimbook-Team/slimbookbattery #112: "Limit CPU profile" setting doesn't seem to apply

byteManiak, 2023-07-04:
**Describe the bug**
When switching from "none" to "maximum" in "Limit CPU profile", `/sys/devices/system/cpu/cpu0/cpufreq/energy_performance_preference` should be set to `power`, but it's still set to `performance`

**To Reproduce**
1. Set "Limit CPU profile" to "none".
2. Set "Limit CPU profile" back to "maximum".

**Desktop (please complete the following information):**
 - OS: Arch Linux
 - Version: latest Linux kernel as of the ticket (kernel 6.4.1)
 - Desktop Environment: KDE Plasma

**Additional context**
I manually set `energy_performance_preference` to `power`:
```
[root@Lenny policy0]# echo power > energy_performance_preference 
[root@Lenny policy0]# cat energy_performance_preference 
power
```
Then I set the Slimbook Battery profile to Maximum performance, which also changes "Limit CPU profile" to "none". This sets `energy_performance_preference` to `performance`, as expected. I also can't write `power` to this while Maximum performance is enabled:
```
[root@Lenny policy0]# echo power > energy_performance_preference 
bash: echo: write error: Device or resource busy
```
Then I set the Slimbook Battery profile back to Energy Saving, which changes "Limit CPU profile" back to "maximum". This should set `energy_performance_preference` to `power`, but it doesn't:
```
[root@Lenny policy0]# cat energy_performance_preference 
performance
```
I can manually set the setting to `power` just fine though:
```
[root@Lenny policy0]# echo power > energy_performance_preference 
[root@Lenny policy0]# cat energy_performance_preference 
power
```
I'm not too sure if this is a Slimbook Battery bug, or if it's something in TLP.

siddhpant, 2023-07-04:
Might be related to https://github.com/linrunner/TLP/issues/630.

utrack, 2023-07-14:
It's definitely related to the TLP issue; the EPP mechanism is new.
EPP is set to performance automatically by the EPP driver when the CPU policy is set to performance (which is why you can't write anything there); however, it does not pin 'power' to powersave or any other CPU policies.

So in short, EPP is not supported yet by TLP or sbb at all, the 'performance' EPP switch is a side effect of the CPU driver.