Slimbook-Team/slimbookbattery #115: i3wm cannot open gui

newyorkthink, 2023-07-26:
[image]
[image]

i run all command ,but is not work

newyorkthink, 2023-07-26:
[image]
i not run . is not icon

siddhpant, 2023-07-26:
Try killing with `$ pkill slimbookbattery` and try running `$ slimbookbattery`.

newyorkthink, 2023-07-26:
> Try killing with `$ pkill slimbookbattery` and try running `$ slimbookbattery`.

i try but is not work

siddhpant, 2023-07-26:
"Does not work" does not help. It's a Python script, so errors will be dumped on the terminal, which you should post by copying and pasting.

Also, did you clone the repo and install, or used the packages? The packages are outdated.

newyorkthink, 2023-07-26:
[image]
i use this

newyorkthink, 2023-07-26:
please update new version

newyorkthink, 2023-07-26:
[image]

newyorkthink, 2023-07-26:
not work

siddhpant, 2023-07-26:
> or can you make appimage format

There is no need, it's just a python script. One can install by cloning the repo and running `deploy.sh`.

siddhpant, 2023-07-26:
> [image]
> 

Hmm, so it keeps running in background? What does `ps` say? I don't know much about the internal working yet, I suppose it is running and the icon doesn't show up on i3? What about GNOME? 

Otherwise you probably could disable the autostart / service and try again after reboot.

newyorkthink, 2023-07-27:
> > [image]
> 
> Hmm, so it keeps running in background? What does `ps` say? I don't know much about the internal working yet, I suppose it is running and the icon doesn't show up on i3? What about GNOME?
> 
> Otherwise you probably could disable the autostart / service and try again after reboot.

i try everything, is not runing in background , but is always show is in execution , and i try everything, the icon is not show in i3wm

siddhpant, 2023-07-27:
Did you update to the latest version? There was a bug fix for pgrep which was merged after 4.0.8.

siddhpant, 2023-07-27:
#110 describes the same problem, and the same thing is mentioned — use the latest version from git. Please search issues before opening a new one.

MarSlimbook, 2023-07-29:
Here is the release [v4.0.9beta](https://github.com/Slimbook-Team/slimbookbattery/releases/tag/v4.0.9beta)

closed 2023-07-29 by MarSlimbook

MarSlimbook, 2023-07-29:
If you want a quick fix for arch this is the commit that should fix the issue:
https://github.com/Slimbook-Team/slimbookbattery/commit/228365ec3cc71816c3bc20d3e3b3b41379e779f3
Just edit /usr/bin/slimbookbattery and correct the pgrep command call

ClashC, 2024-09-25:
> Here is the release [v4.0.9beta](https://github.com/Slimbook-Team/slimbookbattery/releases/tag/v4.0.9beta) I'll close this issue and keep the original one.

Thank you very much! For info, this version works here compared to the 4.0.8beta on Linux Mint 22