Slimbook-Team/slimbookbattery #118: please build appimage format at slimbookbatttery or update new version to repository

newyorkthink, 2023-08-11:
fist  use the following command is break local system package dependence

sudo apt-get install ./slimbookbattery

so, use APPIAMGe format it can run every
and other option you team please update new version to you official repositiory

and people install it , not have show this is obsolete package

because when i  use apt command clean system, they always show the dependant package is obsolete

siddhpant, 2023-08-11:
The previous issue says the same thing, which must have been at the top when you opened the issues page.

I'm not the maintainer, but I have to ask — why create a duplicate for the same issue and create noise for the maintainer?

newyorkthink, 2023-08-12:
> The previous issue (#117) says a similar thing, which must have been at the top when you opened the issues page. Also, #109 says the same thing.
> 
> I'm not the maintainer, but I have to ask — why create a duplicate for the same issue and create noise for the maintainer?
> 
> Edit: I just realised you created #109.
> 
> Anyways, check #118 discussion. And I would suggest to close the duplicates.

I just forgot my issue