Slimbook-Team/slimbookbattery #125: The toggle of "disable animations" doesn't work when changing the profile from the system tray icon

pedroluislopez, 2024-01-04:
**Describe the bug**
The toggle of "disable animations" doesn't work when changing the profile from the system tray icon.

**To Reproduce**
Steps to reproduce the behavior:
1. Change the profile in the system tray icon to "powersave".
2. The animations continue active.
3. Go to "Advance mode" and click on the button "Accept".
4. Now the animations are disabled.

**Desktop (please complete the following information):**
 - OS: Ubuntu
 - Version: 23.10
 - Desktop Environment: Gnome Shell 45.1 - Wayland

**Expected behavior (Optional)**
The configuration of "disable animations" should work when you change the profile from the tray icon.

kzhe2059, 2024-01-06:
I have the opposite issue where I can't change any settings including making all profiles keep animations

kzhe2059, 2024-01-06:
Nevermind. Apparently the label on the toggle is mistranslated and the toggle is inverted