Slimbook-Team/slimbookbattery #137: Linux Mint Debian Edition: Slimbookbattery does not launch

DutchPete, 2024-12-14:
**Describe the bug**
Clicking on the icon in the main menu does nothing.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to [install package on opensuse](https://software.opensuse.org//download.html?project=home%3ASlimbook%3ASlimbookBattery&package=slimbookbattery)
2. install using the 4 commands shown
3. open LMDE's main menu
4. scroll to Slimbookbattery icon, then click on it

**Desktop (please complete the following information):**
 - OS: Linux Mint Debian Edition
 - Version: 6 (Faye)
 - Desktop Environment: Cinnamon

**Expected behavior (Optional)**
The app starts and its popup is shown

tomicakorac, 2024-12-29:
Same issue on Ubuntu Linux 24.10 Oracular, Gnome environment.

c0d3r9, 2025-08-07:
After starting the icon in the taskbar is closing after a few seconds Thats all.
OS: Debian 13 Trixie KDE (Wayland)